commaai/openpilot #20057: Alert message to immediately connect to the internet for updates even though OP is connected

pjankovi, 2021-02-10:
**Describe the bug**

This issue just started happening yesterday: after the drive, the OP displays the following message even though the C2 is connected normally to the internet:

 "Immediately connect to the internet to check for updates. If you do not connect to the internet, openpilot won't engage in 0 days."

So the OP does not engage anymore.  The OP however does uploads drives normally as well. 

****
This happens every time after I turn off the car after my drive.

**Expected behavior**

This should not be happening when it's clear that the OP is connected to the internet. 

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: ffccc77938ddbc44
 - Route: ffccc77938ddbc44|2021-02-09--14-31-24
 - Timestamp: at the end when I turn off the car
 - Version: master-ci
 - Car make/model: 2021 Toyota Camry Hybrid LE

adeebshihadeh, 2021-02-12:
Your device isn't on msater-ci, it's on an old branch based off master. I'd recommend reinstalling master-ci then switching to release2 after 0.8.2 is in release.

closed 2021-02-12 by adeebshihadeh

pjankovi, 2021-02-12:
Thanks! So if I enter the following, would I accomplish what you are suggesting?

cd /data && rm -rf openpilot && git clone -b master-ci https://github.com/commaai/openpilot && reboot

If this is not doing what is supposed, then I am not sure how to reinstall master-ci. So I would appreciate if you can send the proper command. Thanks!

adeebshihadeh, 2021-02-12:
Yes, that should work.