commaai/openpilot #20072: Centralize definitions of mathematical and physical constants

wolterhv, 2021-02-13:
Currently, I've seen π or some approximation to it be defined in 5 files. This can be shown with the output of `grep -r '3\.14' $OPENPILOT_TOPLEVEL`:

```
./selfdrive/car/ford/carcontroller.py:        curvature = self.vehicle_model.calc_curvature(actuators.steerAngle*3.1415/180., CS.out.vEgo)
./selfdrive/car/mock/interface.py:LPG = 2 * 3.1415 * YAW_FR * TS / (1 + 2 * 3.1415 * YAW_FR * TS)
./selfdrive/controls/lib/lateral_mpc/generator.cpp:#define PI 3.1415926536
./selfdrive/modeld/thneed/include/msm_kgsl.h: * are cherry picked into other trees out of order so it is frozen as of 3.14.
./selfdrive/locationd/ublox_msg.cc:      double gpsPi = 3.1415926535898;
./selfdrive/locationd/test/ephemeris.py:    gpsPi = 3.1415926535898
```

Something similar happens with the gravitational acceleration constant g (often defined as 9.81 or even as 10) and probably with other constants as well. Naturally, this occurs in python source code as much as in c++ code.

I propose that all mathematical and physical constants be defined in a central location. It may be defined in a language-agnostic format and have language-specific files generated at build time for inclusion in source code, e.g. with m4. My proposal is that this file be located in `common/constants.txt` and the make procedure can generate therefrom `constants.py` and `constants.h`.

What are other opinions in regard to this? It's not a big problem, probably not a problem at all -- I don't think it could be causing any bugs. It can be solved pretty effortlessly though, and I'm willing to implement it.

pd0wm, 2021-02-19:
Don't think reading from a configuration file will make things easier. But we can definitely clean up all the definitions of pi and G.

wolterhv, 2021-02-19:
I didn't mean reading from a configuration file. I'll explain what I had in mind of generating language-specific source files with m4.

Suppose you wanted to define three constants: pi, Earth's gravitational acceleration and the Earth's radius. Your codebase is in multiple languages and you don't want to maintain more than one file. So you use m4 and create the following file, `constants.m4`:

```m4
dnl Macros
define(`constdef',`ifdef(`cpp',`#define $1 $2',`$1 = $2')')dnl
define(`comment',`ifdef(`cpp',`// $1',`# $1')')dnl
dnl File headers
ifdef(`cpp',dnl
`// This is our automatically generated constants file',
`#!/bin/env python3
# This python script is automatically generated')

comment(`Mathematical constants')
constdef(`PI',`3.14159')

comment(`Physical constants')
constdef(`GRAV',`9.81')

comment(`Geographical characteristic numbers')
constdef(`EARTH_RADIUS',`6378100')
```

which then, at build time, can be used to generate a python version as well as a cpp version.

To generate the python version, you would simply run `m4 constants.m4`, and this command will output the following text:

```python
#!/bin/env python3
# This python script is automatically generated

# Mathematical constants
PI = 3.14159

# Physical constants
GRAV = 9.81

# Geographical characteristic numbers
EARTH_RADIUS = 6378100
```

whereas running instead `m4 -Dcpp constants.m4` would output

```cpp
// This is our automatically generated constants file

// Mathematical constants
#define PI 3.14159

// Physical constants
#define GRAV 9.81

// Geographical characteristic numbers
#define EARTH_RADIUS 6378100
```

That's all there is to it. I notice that m4 is not used in this project, so I anticipate opposition to adding a new tool to the tool chain, but as demonstrated, it can be very useful.

pd0wm, 2021-02-22:
If we had tens or hundreds of physical constants it would probably make sense to add something like that. But don't think it's worth the complexity.

I would merge something that replaces all the instances of PI with `math.pi` and `M_PI` though!

closed 2021-02-28 by adeebshihadeh

wolterhv, 2021-03-02:
I'm not sure how to build the project yet but these changes shouldn't break anything. I'm only concerned about the compilation of `generator.cpp` since it didn't previously include `math.h` and now I've added that inclusion.

You can see the commit with the changes at https://github.com/commaai/openpilot/compare/master...wolterhv:pi_replace.

I'll try to figure out tomorrow how to build the project and test these changes before submitting a PR.

pd0wm, 2021-03-03:
Just checked with @HaraldSchafer, but the gpsPi is actually defined by the GPS standard, so we can't use the regular PI for that.

wolterhv, 2021-03-03:
Oh, I had a feeling about that one. Thanks for letting me know. I've updated my branch, I'll make a pull request.