commaai/openpilot #20075: Define locations for file storage in a configuration file, instead of using python's __file__ path

wolterhv, 2021-02-13:
Currently, there are about 17 lines spread among different files in the openpilot codebase where something similar to the following occurs:

```python
SOME_DIR = os.path.dirname(os.path.abspath(__file__))
```

which is merely a definition of a path, probably for storing files or reading previously stored ones. The benefit of this is probably being able to define paths dynamically, without breaking the system if something is moved from one place to another. 

There is however a cleaner way to do this, which is defining these locations in a configuration file instead, like an INI file or a YAML file. Then all the algorithms which need access to these paths will be able to get them by reading the configuration file. There are plenty of lightweight configuration-file-reading libraries out there which could be used to accomplish this.

I'm interested in hearing thoughts on this enhancement request.

adeebshihadeh, 2021-02-13:
This doesn't sound cleaner, but maybe I'm misunderstanding what you mean. Open a PR with an example if you'd like to discuss further.

closed 2021-02-13 by adeebshihadeh

wolterhv, 2021-02-14:
Well yes, I guess "cleaner" is also a matter of opinion. I have created a very rough PR to better illustrate what I propose. Let me know what you think.