commaai/openpilot #20076: Warning that device is not able to connect to the internet, does not go away regardless of 4G or WiFi. Problem remains even after reboot. 
No change to prime subscription and still able to view car from app.


https://photos.app.goo.gl/sEAkDVhLuS7UahcW8

https://photos.app.goo.gl/oDMDdVtnx35UzFMf7

justinmerrell, 2021-02-13:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

adeebshihadeh, 2021-02-13:
Closing due to lack of necessary information. Feel free to reopen after filling out the template.

justinmerrell, 2021-02-13:
Edited.

justinmerrell, 2021-02-13:
> Closing due to lack of necessary information. Feel free to reopen after filling out the template.

Should I make a new report?

adeebshihadeh, 2021-02-13:
Can you ssh in to your device and try running `cd /data/openpilot && git fetch --dry-run`? Paste the output here.

justinmerrell, 2021-02-14:
From https://github.com/commaai/openpilot
 + 872e0c17...0b476aba master-ci        -> origin/master-ci  (forced update)
 * [new branch]        agnos06          -> origin/agnos06
   40919a6f..c189d15a  master           -> origin/master
 + 698caf46...2440b640 new-model        -> origin/new-model  (forced update)
   ddf0aebb..d229b989  pandad-refactor  -> origin/pandad-refactor
 + b2b986c9...1f51d030 release3-staging -> origin/release3-staging  (forced update)
 * [new branch]        tesla-ap2        -> origin/tesla-ap2
 + 9a976fe3...d3eb7b9a testing-closet   -> origin/testing-closet  (forced update)
   e08b0b47..6c309220  tici_db_remaster -> origin/tici_db_remaster
 * [new branch]        ui-fixes         -> origin/ui-fixes
 + b2383d91...00490bae zookeeper-ci     -> origin/zookeeper-ci  (forced update)

adeebshihadeh, 2021-04-27:
I checked your device's logs over the last week and seems it's updating fine now, so we'll close this. I also fixed up the error handling in #20754, so the exceptions you would've been getting will show as an alert and in the logs for debugging.

closed 2021-04-27 by adeebshihadeh