commaai/openpilot #20082: SCons build fails with missing module compilation_db

NielsPichon, 2021-02-15:
**Describe the bug**
When trying to build  openpilot using `scons` in the home directory of  openpilot, it fails with the following error message: 

SConsEnvironmentError: No module named compilation_db:
  File ".../openpilot/SConstruct", line 147:
    env = Environment(
  File "/usr/lib/scons/SCons/Environment.py", line 990:
    apply_tools(self, tools, toolpath)
  File "/usr/lib/scons/SCons/Environment.py", line 102:
    env.Tool(tool)
  File "/usr/lib/scons/SCons/Environment.py", line 1809:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 122:
    module = self._tool_module()
  File "/usr/lib/scons/SCons/Tool/__init__.py", line 220:
    error_string = "No module named %s" % self.name

Looking in `/usr/lib/scons/SCons/Tool/__init__.py` where the eroor was thrown, it seems like scons is looking for the following file `./openpilot/site_scons/site_tools/compilation_db.py'`. In the `site_scons/site_tools` directory I only have `cython.py`. Is that expected? Am I missing some setup steps? 

Thanks for your help :slight_smile:

**How to reproduce**

```
cd $HOME/openpilot
scons -j$(nproc)
```
I tried running `ubuntu_setup.sh` script in the tools directory in case I was missing some dependencies and it did not help.

**Device/Version information (please complete the following information):**
 - Build on Ubuntu 20.04LTS
 - Version: c189d15af9a613d8f109b39298c0ab3e22f39f6d on master branch

pd0wm, 2021-02-15:
Looks like you're using scons from the ubuntu repos. Make sure you use the latest one using `pipenv`.

adeebshihadeh, 2021-02-15:
Compilation database support is new in scons as of v4.0. Update scons or use the version in our Pipfile.

closed 2021-02-15 by adeebshihadeh

NielsPichon, 2021-02-16:
Thanks both. It seems this error message is indeed gone now.