commaai/openpilot #20084: Calculate vehicle speed from non-driven axle only.

turnbeutelvergesser, 2021-02-15:
Vehicle speed estimation can be slightly improved if you don't average all four wheel speed sensors but only the ones of the un-driven axle, because then you don't let ire slip degrade the estimation. See i.e. https://www.hondata.com/help/tractioncontrol/theory_of_operation.htm.

https://github.com/commaai/openpilot/blob/34ff2951d4172710c35f0443f2a4a274a1b73d70/selfdrive/car/volkswagen/carstate.py#L24

Golf and Audi A3 are front-wheel- driven.

grekiki, 2021-02-15:
Wondering if the difference would even be significant on the highway. So if I understand the link correctly you'd notice the most difference under high acceleration when it could even achieve 1-3% due to tyre slip because of the engine torque on the wheels. Sounds like something to look at in the data, as to how much difference there is (if it is larger than noise). 

I am worried about it underestimating the speed of the center of car in the corners though, since the rear wheels take the shorter path.

grekiki, 2021-02-16:
Did some plotjuggler plotting. 1-2 % speed offset is quite reproducible, so we may miss our speed average by about 1%. However just looking at acceleration would be problematic due to a potential acceleration in the corner, where the offset between front and rear wheels would be much larger than 1-2% ( 10% is easily achieved in tight corners). The way to fix this problem is to, at some point, use the speeds from the model/localizer which shouldn't depend on wheel speeds. 
So this definitely is an issue, but I don't think that taking the speed of the rear wheels is an improvement due to the potentially bad estimates in corners.

grekiki, 2021-02-16:
[image]

closed 2021-02-16 by grekiki