commaai/openpilot #20086: TSS2 cars need angle offset for more accurate angle signal from torque message

ShaneSmiskol, 2021-02-16:
This line seems to be incorrect along with the comment: https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/carstate.py#L22

On birdman's TSS2 Corolla that `STEER_TORQUE_SENSOR` `STEER_ANGLE` signal that's being used always seems to start at 0, so the angle openpilot uses can have an offset that varies wildly between drives depending on where the wheel starts, since it's not being offset when it should.

This is simulating what carstate does now (but doesn't actually offset it, so you can see the difference in angles. It should have been offset -2.95829... degrees.
[image]

Here's also the different angles plotted against each other:
[image]

From carstate is what openpilot uses, and from can is the message available and used on TSS1 cars that's not as precise (`STEER_ANGLE_SENSOR`, `STEER_ANGLE`).

ShaneSmiskol, 2021-02-16:
Considering my TSS1 Corolla's offset is about -0.31 degrees this is rather high which could cause some weirdness on initial engage as openpilot works to find out the offset that carstate should have taken care of: 
[image]

pd0wm, 2021-02-16:
Are you suggesting to replace line 22 by `self.needs_angle_offset = True`. I guess that can't hurt.

ShaneSmiskol, 2021-02-16:
Yeah, if you found nodsu non-TSS2 cars need the offset too then I bet they all do with the more accurate sensor. If a specific car doesn't need the offset then both angles will start at the same value keeping the offset 0 hopefully

closed 2021-02-16 by pd0wm