commaai/openpilot #20109: "CAN Error: Check connections" -- Not related to cable connections

mhosks, 2021-02-18:
**Describe the bug**

"CAN Error: Check connections" shows up on a regular basis for a reason not related to loose connections. This issue typically resolves when the car is off for 2+ hours with no adjustment to any connections.

**How to reproduce or log data**

It seemed like it was happening following a long drive, shutting the car off for 15+ minutes, and the error would appear with car start. However, "CAN Error" showed up after first starting the car this morning.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 8cf22cabbda6230e
 - Route: This particular route is suspicious because of the short distance so I need to upload other routes from today [Feb. 18]. 8cf22cabbda6230e|2021-02-18--11-04-22
 - Timestamp: Whole drive
 - Version: 0.8.2 release2
 - Car make/model: Toyota Corolla 2021

pd0wm, 2021-02-19:
Can you make sure the route is fully uploaded? Please connect to wifi and check `https://my.comma.ai/useradmin/?onebox=8cf22cabbda6230e%7C2021-02-18--11-04-22` to make sure the rlogs are present.

mhosks, 2021-02-21:
I uploaded some rlogs. No idea if that's all of them. The way I did it was somewhat finnicky.

I have another route with CAN Error. Again, not related to cable connections because the error disappeared on the next car start. Here it is: `8cf22cabbda6230e|2021-02-20--12-31-12`

adeebshihadeh, 2021-02-21:
In the routes with the CAN errors, you're missing messages on bus 2. Check your harness connections and ensure all the wires are in good condition. If the issue persists, contact support@comma.ai.

closed 2021-02-21 by adeebshihadeh