commaai/openpilot #20113: Hyundai Kona EV 2019, Collision Sensor error message.

mstoteler, 2021-02-19:
When engaging open pilot, the car gives an 
error message to check the forward collision sensor. And gives a few orange error symbols on the car dashboard. (see attached image)

Device: Comma 2
Dongle ID: 9e02651652ef6c6f
Software version: Master-Ci (0.8.1)

Car make/model: HYUNDAI KONA EV 2019
Route: 9e02651652ef6c6f|2021-02-19--07-48-20

[image]

adeebshihadeh, 2021-02-28:
Has this always happened on your car or is it a recent regression?

mstoteler, 2021-03-01:
I only know that it is the case now. The FCA warning always shows op when engaging OP.

adeebshihadeh, 2021-03-11:
Can you record a route with the stock system and post it here? All you need to do is disable openpilot in the settings before starting your car. No need to drive, just leave your car on for a minute or so.

mstoteler, 2021-03-11:
I recorded a drive, disabled openpilot and on stock OP on release2.

9e02651652ef6c6f|2021-03-11--12-02-15

adeebshihadeh, 2021-03-11:
Can you test this branch? https://github.com/commaai/openpilot/pull/20320

The discord community (https://discord.comma.ai) can help you with checking out the branch.

mstoteler, 2021-03-12:
I've tested it this morning. But still the same message..

9e02651652ef6c6f|2021-03-12--08-56-28

mstoteler, 2021-06-01:
Is this problem not fixable? My car doesn't emergency break with OpenPilot engaged?

adeebshihadeh, 2021-06-02:
This problem is certainly fixable, but your car is a community maintained car, meaning support largely comes from the community. Yes, emergency braking is likely affected given the fault you're seeing.

sshane, 2022-07-07:
Is this still an issue on stock openpilot, but not a fork?

sshane, 2022-07-12:
@xx979xx Do you happen to remember if any of your changes fix this user's issue?

DerGue69, 2022-07-20:
Hi. I have Kia Niro MY 19 - EU Austria 1st Edition. I am facing the same problem. Yesterday  I installed stock OpenPilot with fingerprint modifications. 
When I engage Openpilot I instantly get the FCA Warning message.

Can you give me a hint what to test, which fork I should try?

Günther

JaredReabow, 2022-07-20:
I may be able to suggest alternatives you can fix your issue using, email me @ ox141jf@gmail.com

DerGue69, 2022-07-21:
> I may be able to suggest alternatives you can fix your issue using, email me @ [ox141jf@gmail.com](mailto:ox141jf@gmail.com)

I sent you an eMail from my gmail!

JaredReabow, 2022-07-21:
Sorry I see no email from you

DerGue69, 2022-07-21:
I sent sent now several mails, from several accounts. If you did not receive anything by now, let me know.

sshane, 2022-07-21:
@mstoteler if you're still using a comma two, please try this branch and let me know how it goes: https://github.com/commaai/openpilot/commits/fix-lfa-warn-c2

sshane, 2022-08-01:
@mstoteler I pushed an update to the branch, let me know if you try it. If you no longer have this issue let me know and I can close this issue,.

sshane, 2023-08-07:
Does this still happen? If so please provide a recent route. @JaredReabow @mstoteler

closed 2023-08-07 by sshane

JaredReabow, 2023-08-07:
I believe that Jyoung found the issue and submitted a fix, however, i
cannot test it right now as my comma 3 is faulty

sshane, 2023-08-07:
@jyoung8607 do you know what it was?

DerGue69, 2023-08-08:
I can test, when you tell me what to do!

DerGue69, 2023-08-08:
Then it is not resolved.
I have a personal  fork, which is a 1:1 clone from master.
Exept the I add the lines in hyundaican.py to hide the error.
Shane said, this hides the error, but we don't know if the FCA within the car is still working.

.........
# Some cars show an FCA fault when forwarding CF_Lkas_FusionState. Hide FCA Fault:
    values["CF_Lkas_FusionState"] = 0
........

Occasionally I remove the lines to see, if I still get the FCA error.
And I still have it on Master 0.9.5

https://github.com/DerGue69/openpilot/blob/master/selfdrive/car/hyundai/hyundaican.py

JaredReabow, 2023-08-08:
If these are the same lines that sunnypilot uses then yes FCA continues to
function, ask me how I know ;p

DerGue69, 2023-08-08:
Unfortunately this line of code is not in sunny fork.
I tried sunny > 1 year ago. And I had the same FCA error by then.

After my vacation I will try Sunny Fork once again.

sshane, 2023-08-08:
I closed since the original user is no longer active here, but we can move discussion to the open issue @DerGue69 has: https://github.com/commaai/openpilot/issues/25221

jyoung8607, 2023-08-15:
> @jyoung8607 do you know what it was?

I'm afraid not. I remember looking into it, but I don't remember arriving at a solution.