commaai/openpilot #20124: Lanes mismarked among old lanes in construction zone

Jon8RFC, 2021-02-21:
**Describe the bug**

Lanes not marked correctly in a construction zone when lanes transitioned from normal to splitting into new lane markers and scrapes where old lanes were.
Significant steering corrections necessary, twice.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 6363a0e48f515f27
 - Route: 6363a0e48f515f27|2021-02-20--15-00-54
 - Timestamp: between 4:04 and 4:06pm
 - Version: 0.8.1
 - Car make/model: Hyundai Sonata 2018

**Additional context**

The clock on my dash said 4:04, but I don't know where specifically the major corrections occurred within 2 minutes of that because I don't know how to view better quality video in cabana from the HQ video uploaded.  Please inform me of how to do so, so that I can give precise information in the future.

doktorsleepelss, 2021-02-23:
In explorer, click on download video segment to get the full res video. View it with VLC.  Can't view the the HQ video inside explorer or cabana.

HaraldSchafer, 2021-04-21:
Our laneline detector does not deal well with splitting lanelines. Laneless policy should deal with this better, if not already it will soon. Once laneless is good enough it will be the default policy.

closed 2021-04-21 by HaraldSchafer