commaai/openpilot #20125: Subaru New MPC issues

michaelhonan, 2021-02-22:
**Describe the bug**

It seems that Subaru models have trouble in 082 with the new MPC changes. openpilot steers quite lazily, and will leave lanes, get super close to lane lines, and also let the car drift towards the curb.

openpilot also no longer appears to care about remaining centered, it will swerve slowly all over the road.

This behaviour was not present in the previous release (0.8.1).

**How to reproduce or log data**

Just simply drive, the car just doesn't steer well. This is consistent on almost every road I drive on.

I'm not sure how to refine this section.

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 4d70bc5e608678be
 - Route: 4d70bc5e608678be|2021-02-22--12-33-49--8 (almost drove into the curb)
 - Timestamp: Mon, 22 Feb, 2021 @ 12:42:07
 - Version: 0.8.2 Personal fork (minor modifications that do not affect this area) based on commit: [c747654](https://github.com/commaai/openpilot/commit/c747654f057fde3adb5dda3a8a032e24b110735c)
 - Car make/model: Subaru Impreza 2018

**Additional context**

Here's another example where openpilot decides to leave a lane: 4d70bc5e608678be|2021-02-22--12-52-24--1 

I will continue to update this issue with more routes as I come across them.

HaraldSchafer, 2021-02-22:
Working on a fix here

https://github.com/commaai/openpilot/pull/20133

michaelhonan, 2021-02-23:
PR #20133 has addressed this issue. Thanks!

RCoaster, 2021-02-24:
The fix didn't help much with the base 0.8.2 release2-staging branch. It still steers lazily around, but stays in the lane. A lot of ping ponging. It's like it's very slow to react to going off center. It still hugs the curves. I have a 2019 Subaru Ascent Limited.

HaraldSchafer, 2021-02-24:
@RCoaster Do you have a route?

RCoaster, 2021-02-24:
Yes: b75e98e3b7cf69dd|2021-02-24--03-11-28

HaraldSchafer, 2021-02-24:
Hmmm, yes I see some bad ping pong. Your car looks badly tuned. See in the plot how it never reaches the desired steering angles? Also my tests indicate the steerActuatorDelay is more like .4+ seconds, not .3seconds. 

Unfortunately, we don't really have the infrastructure to autotune cars yet and we don't have any subarus at the moment so we can't really test on them :/.

[image]

michaelhonan, 2021-03-01:
Seems more Subaru people are having trouble with this. Reopening.

RCoaster, 2021-03-01:
Here are the 4 segments of a 400 mile road trip I did today. This is a 2019 Subaru Ascent. 
b75e98e3b7cf69dd|2021-02-28--08-35-53 
b75e98e3b7cf69dd|2021-02-28--08-54-41
b75e98e3b7cf69dd|2021-02-28--10-56-18
b75e98e3b7cf69dd|2021-02-28--13-25-09 (This segment did not have any noticeable ping-ponging, and about 2:05:12 (while passing the tractor trailer) the set cruise speed corrected to the correct set speed- it was set for 75 mph and it would show 47 mph. When there's an update, it goes back showing the wrong mph number).

michaelhonan, 2021-03-05:
Closing, since it seems to be as a result of tuning.

closed 2021-03-05 by michaelhonan