commaai/openpilot #20127: light_sensor value sticking

spektor56, 2021-02-22:
**Describe the bug**

The brightness on my device is not adapting consistently when covering the light sensor with my thumb during the day or shining a light at it during the night.  I noticed at night when I shine a light at the light sensor then move it away that the value for s->light_sensor is getting stuck at a high value and not returning to 0 in the dark.  To get it to move to 0 I need to keep moving my light over the sensor to trigger the sensor events.

in ui.cc:
float clipped_brightness = fmin(512, (s->light_sensor*brightness_m) + brightness_b);
smooth_brightness = fmin(255, clipped_brightness * 0.01 + smooth_brightness * 0.99);
ui_set_brightness(s, (int)smooth_brightness);

**How to reproduce or log data**

In a dark room, shine cell phone light at the light sensor
In a bright room, cover the cell phone light sensor with your thumb

check the value of s->light_sensor

**Expected behavior**

the brightness of the screen should adapt to the environment

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Version: 0.8.1-release

**Additional context**

adeebshihadeh, 2021-02-24:
This is likely a hardware issue. Contact support@comma.ai and we can swap it out for you.

spektor56, 2021-02-25:
I dont think its a sensor issue.  When the screen is off the light sensor value updates very quickly and doesn't stick.  when the screen comes on and the UI shows up the value updates much more slowly and gets stuck sometimes

cydia2020, 2021-03-04:
Have seen this a few times before as well, I once went off-road and used my phone's flashlight in the vehicle's cabinet. The c2's brightness got stuck and never dimmed, had to restart the device to offset this behaviour, afterwards it seems to function correctly, this was on 080 though.

adeebshihadeh, 2021-03-09:
Ah, this is because the light sensor values are relatively low frequency, and UI subscribes to sensorEvents with a socket that has conflate enabled. This means that UI misses a lot of light sensor values because the other sensors update 10x more frequently. This will be fixed in the sensor refactor we'll be doing soon.

closed 2021-03-09 by adeebshihadeh