commaai/openpilot #20135: Intermittent "Car Unrecognized"  toyota highlander 2021

jaenmora, 2021-02-23:
Intermittent "Car Unrecognized"  toyota highlander 2021

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

vanillagorillaa, 2021-02-23:
Please edit your original post and provide the relevant information (I.E. DongleID)

VirtuallyChris, 2021-02-23:
Your car had missing engine fw. I've added it in this PR https://github.com/commaai/openpilot/pull/20144

master-ci will have this in a few hours.

closed 2021-02-23 by VirtuallyChris