commaai/openpilot #20158: The vehicle can not run properly in the simulator

LLplane, 2021-02-25:
The vehicle can not run properly in the simulator and always  always hit the edge of the road.

below are some screenshots 
[image]

the car hit the edge
[image]

Operating system: [e.g. Ubuntu 18.04]

westbrook00, 2021-02-25:
[image]
Hello,LLplane! Have you got this error before? I just follow the steps:1. ./start_carla.sh 2../start_openpilot_docker.sh, then it can't find the attribute "angelSteersDes", if anyone helps me fix the error,I will thank a lot!

adeebshihadeh, 2021-02-28:
Closing this as there are several problems and open issues for the simulator. We'll be spending some time on getting the simulator in a good state and fixing the open issues very soon.

closed 2021-02-28 by adeebshihadeh