commaai/openpilot #20170: display fingerprinted car in settings

adhintz, 2021-02-26:
When debugging OpenPilot issues, it'd be nice to have an easy way to view what car it's fingerprinted as. Previously I've been SSHing in and running tmux.

Steps to add:

1. In selfdrive/car/car_helpers.py get_car do `Params().put("CarFingerprint", candidate)`
Optionally include the FingerprintSource to differentiate between can, fw, or fixed fingerprinting method.

1. Add CarFingerprint to common/params_pyx.pyx

1. In selfdrive/ui/qt/offroad/settings.cc developer_panel read in `params.get("CarFingerprint", false)`

Sound good? If so, I can implement it.

adeebshihadeh, 2021-02-28:
I think this may be more confusing than helpful for most users. For example, "TOYOTA COROLLA 2017" also covers the 2018-2019 years. I also don't think misidentifying a car is a common issue.

adhintz, 2021-02-28:
When helping people remotely, I typically ask them what it's fingerprinting as. I don't recall how often it's been an issue, but perhaps the most common is when it doesn't fingerprint as anything. I remember this happening for a new year that sends a few messages with different lengths.

Adding it was easy enough: https://github.com/adhintz/openpilot/commit/bafe547d8b26e199d3bfc3b66b217ea8e0fc3571

However then the name is a bit long and could lead to confusion as you mentioned:
[image]

Perhaps it would be useful to display just the detected car's manufacturer, that no fingerprint matched, or that there was no CAN traffic? If not, of course feel free to close this.

adeebshihadeh, 2021-02-28:
I'm still not a fan of an extra label in the settings, but I agree it's confusing when a car doesn't fingerprint at all. We show a different startup alert ("Car Unrecognized"), but it's easy to miss that. We'd certainly consider a PR for a good, persistent indicator that you're in dashcam mode. As for identifying this after the fact, it's easy to tell from the logs (all relevant info is in carParams), and we've just added a feature to useradmin to filter by packet type: choose a route from useradmin -> select a segment -> select carParams from the dropdown on top.

adhintz, 2021-02-28:
Sounds good.

closed 2021-02-28 by adhintz

gregjhogan, 2021-03-01:
Does a qlog get get generated in your scenario? If yes, you can select the first segment and see things like this.

adhintz, 2021-03-18:
@adeebshihadeh is useradmin an internal Comma tool? or is there a way I could see the fingerprinted car in a shared Cabana drive?