commaai/openpilot #20201: 0.8.2 breaks Nissan Leaf (no longer recognized)

willemcvu, 2021-03-01:
**Describe the bug**

After installing 0.8.2 and ensuring Community Features is enabled, OpenPilot no longer properly recognizes my 2018 Nissan Leaf, and goes into dashcam-only mode.

**How to reproduce or log data**

1. Install 0.8.2
2. Enable Community Features
3. Try to use on a 2018 Nissan Leaf.

**Expected behavior**

In 0.8.1 it worked fine.

**Device/Version information (please complete the following information):**
 - Comma 2
 - Dongle ID: e4d79cf6b8b19a0d
 - Route: e4d79cf6b8b19a0d|2021-02-28--08-56-44
 - Timestamp: Whole time
 - Version: 0.8.2
 - Car make/model: 2018 Nissan Leaf

willemcvu, 2021-03-01:
Small update after some testing - the issue has been narrowed down to the new `CAR.LEAF_IC` fingerprints. Removing the first first  `CAR.LEAF_IC` fingerprint from `/selfdrive/car/nissan/values.py` makes my 2018 Nissan Leaf come out of dashcam mode.

avolmensky, 2021-03-01:
From discord, this is the fingerprint that @willemcvu has collected from his leaf

```2: 5, 42: 6, 264: 3, 361: 8, 372: 8, 384: 8, 389: 8, 403: 8, 459: 7, 460: 4, 470: 8, 520: 1, 569: 8, 581: 8, 634: 7, 640: 8, 644: 8, 645: 8, 646: 5, 658: 8, 682: 8, 683: 8, 689: 8, 724: 6, 758: 3, 761: 2, 783: 3, 852: 8, 853: 8, 856: 8, 861: 8, 944: 1, 976: 6, 1008: 7, 1011: 7, 1057: 3, 1227: 8, 1228: 8, 1261: 5, 1342: 1, 1354: 8, 1361: 8, 1459: 8, 1477: 8, 1497: 3, 1549: 8, 1573: 6```

adeebshihadeh, 2021-04-05:
Merging the two LEAF_IC fingerprints doesn't fix @willemcvu's case. Since this was a regression for an existing supported car, we'll remove the new, conflicting fingerprint, which I believe is @razem-io's. This specific case can be fixed by finding an alternate seatbelt signal so that we can merge the LEAF and LEAF_IC cars. The longer term fix is to get firmware fingerprinting support for Nissan. Perhaps you're interested in this, @avolmensky? Here's a [recent PR](https://github.com/commaai/openpilot/pull/20271) for VW firmware fingerprinting as an example.

razem-io, 2021-04-05:
Sure, I will just use my fork in the meantime. I am pretty sure that there is no other seatbelt signal. A friend and I did alot of tests to check every signal for correlation programmatically. We did not find anything else.

IanLEAFplus, 2021-04-06:
Can someone explain to a Noob such as myself how I can go about getting my Comma Two functioning again on my LEAF Plus?  I am stuck in dashcam mode and am not sure of how to edit fingerprints, or anything like that.  In need of a YouTube walk through if possible!

adeebshihadeh, 2021-04-06:
Removed the conflicting fingeprint in 90d6bcc8e9fdb12116c1dbd30af2514f514e4e4b. 

@razem-io we could also merge the two if we had a signal to differentiate them. maybe there's a unique CAN message only for the IC model? You could also give implementing firmware fingerprinting a shot. That shouldn't be too hard and would benefit Nissan support in openpilot as a whole.

closed 2021-04-06 by adeebshihadeh

razem-io, 2021-08-01:
Fingerprinting V2 is currently being tested here: https://github.com/razem-io/openpilot/tree/nissan-fpv2-0.8.6