commaai/openpilot #20204: hugging nearside

bitbob1, 2021-03-01:
car tends to hug too close to nearside at certain places.

hugging to nearside. here is an example though it does it at other places too.

**How to reproduce or log data**

expect car to be more central to lane. ie: better lane positioning

comma 2.
 dongle : dc0a54021740121
 route01\03\2021. 3:09 pm - 3:22pm. 
 time stamp: -06:32 - 05:59
 version 0.8.2
 car: toyota prius 2018

whilst running master-ci before 0.8.2 release I noticed big improvements towards the end. The car would not hug at this point on the road and other points too. So somewhere along the line things have changed , unfortunately for my car not so good. bazerk.

davetapley, 2021-03-04:
I've been seeing this in 0.8.2 (compared to 0.8.1) with 2019 Toyota Highlander Hybrid, I'll try and get some example route IDs, but wanted to get a +1 in now.

Richardrgc74, 2021-03-07:
I have also noticed this in the update.. I often have to intervene to stop it riding onto the curb - generally on urban roads.

HaraldSchafer, 2021-04-21:
Unfortunately since we started improving the segnet it would detect curbs as lanelines less, which actually regressed the driving in those scenarios even when the laneline detector actually improved.

Laneless policy should deal with this better, if not already it will soon. Once laneless is good enough it will be the default policy.

closed 2021-04-21 by HaraldSchafer