commaai/openpilot #20206: GPS Malfunction error on 0.8.2

davetapley, 2021-03-01:
**Describe the bug**

On car start Comma 2 will display:
> **GPS Malfunction**
> Contact support

**How to reproduce or log data**

1. Upgrade from 0.8.1 to 0.8.2
1. Stop/start car

Only seen once, and not since (10+ stop/start cycles).

**Expected behavior**

No error 🤪 

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: `2e77efd3816f5347`
 - Route: `2e77efd3816f5347|2021-02-28--16-58-24`
 - Timestamp: At start
 - Version: 0.8.2
 - Car make/model: Toyota Highlander Hybrid 2019 w/ C-SDSU

**Additional context**

Seen reported by someone else in Discord.

Preceding route to one with failure shows `can` `false` in `useradmin`, not sure if related to upgrade to 0.8.2 and/or this issue.

I used `strings <qlog file>  | grep -i gps` and `diff` to compare failure with previous route.
I see this **only** on failure:
```
GPS Malfunction gpsMalfunction/permanent
GpsLocationProvider AGpsStatus is V2+: 216
GpsLocationProvider AGPS IP is v4: 7f
GpsLocationProvider
GpsLocationProvider Passing AGPS IP addr: size 4
GpsLocationProvider
GpsLocationProvider
```

nzyy912, 2021-03-02:
See the same issue on Hyundai Sonata 2021.

vanillagorillaa, 2021-03-02:
@nzyy912 Please edit your post and include your dongleID and a route it occurs on.

nzyy912, 2021-03-02:
Can I ask how do I find these numbers? On the Comma 2?

> @nzyy912 Please edit your post and include your dongleID and a route it occurs on.

vanillagorillaa, 2021-03-02:
Settings -> Device -> DongleID

sjburt, 2021-03-02:
Also had this issue: dongle ID 457d42782c83b77c , route 457d42782c83b77c|2021-03-01--14-50-51

strickc, 2021-03-07:
I had this issue also.  The C2 flashed "GPS Malfunction" and "Camera Manfunction" error messages and my Rav4H threw a "Cruise control malfunction visit dealer" error message.  C2 was not engaged at the time.  It happened both when I arrived and was leaving a park in the forest that has spotty mobile reception and potential GPS obstruction from the trees.  So it may be a legitimate localization error although undesirable for that to cause malfunctions, including CAN errors.
Dongle ID is `a9badec81d3f1943`.
Routes are: `a9badec81d3f1943|2021-03-06--07-57-33` at the end and `a9badec81d3f1943|2021-03-06--09-19-54`

sjburt, 2021-03-07:
I had this issue again, Message "GPS Malfunction Contact support". Restarting the C2 did not help. powering down the C2, unplugging it, and plugging it back in again later did.

dongleID `457d42782c83b77c`, route `457d42782c83b77c|2021-03-07--06-11-22` and `457d42782c83b77c|2021-03-07--06-13-41`

pd0wm, 2021-03-10:
The ublox seems to suddenly lower the message frequency from 10Hz to 2Hz or even 1Hz. Added some logging to boardd, so will turn off the alert until we fix the root cause.

llck80ll, 2021-03-22:
Hi, this is the second time getting the GPS error message since receiving and installing the Comma Two this past friday.

I unplug and plug the device and it fixes the issue.

sjburt, 2021-03-22:
To get the fix from #20278 you will need to install master-ci or wait for the next release.

llck80ll, 2021-03-22:
> To get the fix from #20278 you will need to install master-ci or wait for the next release.

I will wait for the next release. Thanks.

chicman, 2021-04-19:
Same issue, 0.8.2.

pd0wm, 2021-04-19:
0.8.3 with a fix has been release three weeks ago. Please update to the latest version.

closed 2021-04-20 by pd0wm