commaai/openpilot #20221: OpenPilot Momentarily Froze at Intersection Causing Car to Abruptly Phantom Brake

C0mpl3t3N00b, 2021-03-02:
**Describe the bug**

Driving along normally with OpenPilot actively controlling my vehicle when all of the sudden my car just slams its brakes when going across an intersection. I had to disengage and apply gas so that the car behind me would not collide.

**How to reproduce or log data**

Unknown.

**Expected behavior**

When OpenPilot is activated, it should stay activated until the driver disengages or the system should warn the driver that it is about to disengage for whatever reason.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: bdad50b8a1f9fa8e
 - Route: bdad50b8a1f9fa8e|2021-03-01--18-03-02--6
 - Timestamp: bdad50b8a1f9fa8e|2021-03-01--18-03-02--6
 - Version: master-ci
 - Car make/model: Toyota Rav4 Hybrid 2021

**Additional context**

pd0wm, 2021-03-03:
I can't find a disengagement in segment 6 of that drive. Can you confirm you have the right segment?

C0mpl3t3N00b, 2021-03-04:
> I can't find a disengagement in segment 6 of that drive. Can you confirm you have the right segment?

bdad50b8a1f9fa8e|2021-03-01--18-03-02--6 should be the right segment. Even replaying it back in the website there is a glitch in that segment.

pd0wm, 2021-03-05:
You mean you don't see the disengagement? Can you screenshot explorer the moment in the drive where the braking happened?

pd0wm, 2021-10-06:
Closing due to inactivity.

closed 2021-10-06 by pd0wm