commaai/openpilot #20222: Driver Monitor does not detect me

rinterdonato, 2021-03-02:
**Describe the bug**
Just updated to 0.8.2 release2 and as soon as I engage the Comma2, the Driver Distracted error appears.  The only way I can get it to see that I am NOT DISTRACTED is to sit with my face closer to the Comma2 (not my usual driving position - or comfortable).  This did not occur with 0.8.1 release2.  Same car (Acura RDX 2018, with Comma Pedal).

**How to reproduce or log data**

**Expected behavior**
I Expect the driver monitor to see me in my normal driving position (back against seat)

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Comma2
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - bf960626d90a2b97
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - https://my.comma.ai/bf960626d90a2b97/1614711844711/1614712448925
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - 12:13:03
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - release2
 - Car make/model: [e.g. Toyota Prius 2016]
 - 2018 Acura RDX

**Additional context**
I also have the Comma Pedal installed.

VirtuallyChris, 2021-03-02:
You don't have driver recording on so it'd be hard to look into this issue. Can you update it with a route where you are recording front cam?

rinterdonato, 2021-03-03:
Will do. Also, when it started misbehaving, I tried to roll to kegman-081 and got communication errors. Just updated back to stock 0.8.2, release2. I hope it works again. 🤞

Thanks,

R

Rich Interdonato

rinterdonato, 2021-03-03:
Chris,

I just turned on the drive camera upload and tested 0.8.2 again and it works fine. No idea what was going on. Maybe the sun was in is eyes? I will continue to watch it. Please close the bug report.

Thanks for the quick follow up. My last drive file shows it working.

R

Rich Interdonato

________________________________
From: Rich Interdonato <rinterdonato@gmail.com>
Sent: Tuesday, March 2, 2021 5:44:30 PM
To: commaai/openpilot <reply@reply.github.com>
Subject: Re: [commaai/openpilot] Driver Monitor does not detect me (#20222)

Will do. Also, when it started misbehaving, I tried to roll to kegman-081 and got communication errors. Just updated back to stock 0.8.2, release2. I hope it works again. 🤞

Thanks,

R

Rich Interdonato

closed 2021-03-03 by adeebshihadeh