commaai/openpilot #20232: Torque and steering angle rate values wrong for some cars

HaraldSchafer, 2021-03-03:
Ideally we would get from every car:
- Steering angle in degrees
- Steering angle rate in degrees/sec
- Human torque (torque human exerts on steering wheel) in NM
- EPS torque (the torque added by the EPS on the steering wheel) in NM

Human torque and steering angle are used in openpilot, so I assume those are correct. The other two seem to be wrong on many cars. I looked at some data to find the cars on which they are wrong, this isn't necessarily complete, but should give good picture. 

### Steering angle rate problems
**Sign flipped on these cars**
`HONDA ODYSSEY 2018 EX-L 
HONDA FIT 2018 EX 
HONDA RIDGELINE 2017 BLACK EDITION
HONDA CR-V 2016 TOURING
HONDA PILOT 2017 TOURING
HONDA PILOT 2019 ELITE
HONDA HRV 2019 TOURING
ACURA ILX 2016 ACURAWATCH PLUS`

**Strictly positive on these cars**
`HYUNDAI GENESIS 2015-2016
GENESIS G70 2018
HYUNDAI IONIQ HYBRID 2017-2019
HYUNDAI SONATA 2019
KIA NIRO EV 2020
KIA STINGER GT2 2018
HYUNDAI SANTA FE LIMITED 2019
`

** Not set on these cars **
`SUBARU ASCENT LIMITED 2019
SUBARU FORESTER 2019
SUBARU IMPREZA LIMITED 2019
CHEVROLET VOLT PREMIER 2017 
NISSAN ROGUE 2019
NISSAN LEAF 2018
NISSAN X-TRAIL 2017
SUBARU OUTBACK 2015 - 2017`

### EPS Torque problems
** Strictly positive on these cars **
`HONDA ODYSSEY 2018 EX-L
HONDA CR-V 2017 EX 
HONDA FIT 2018 EX 
HONDA ACCORD 2018 LX 1.5T
HONDA RIDGELINE 2017 BLACK EDITION
HONDA ACCORD 2018 SPORT 2T
HONDA ACCORD 2018 HYBRID TOURING
HONDA CIVIC 2016 TOURING
HONDA CIVIC HATCHBACK 2017 SEDAN/COUPE 2019 
HONDA HRV 2019 TOURING
HONDA INSIGHT 2019 TOURING`

** Is not set on these cars **
`HONDA CR-V 2016 TOURING 
SUBARU ASCENT LIMITED 2019
SUBARU FORESTER 2019
SUBARU IMPREZA LIMITED 2019
ACURA RDX 2018 ACURAWATCH PLUS
CHEVROLET VOLT PREMIER 2017
VOLKSWAGEN GOLF
NISSAN ROGUE 2019
ACURA ILX 2016 ACURAWATCH PLUS
NISSAN LEAF 2018
NISSAN X-TRAIL 2017`

ffhspa, 2021-03-04:
Hello @HaraldSchafer 
in your analysis Car Honda Civic Bosch Diesel (CIVIC_BOSCH_DIESEL) is missing 
Here is a DongleID: d83f36766f8012a5 for further analysis from your side. 
Below are some routes: 
d83f36766f8012a5|2021-03-04--08-51-34
d83f36766f8012a5|2021-03-04--08-46-01
d83f36766f8012a5|2021-03-04--08-39-09
d83f36766f8012a5|2021-03-04--07-32-11

Thanks a lot

ffhspa, 2021-03-08:
Hello @HaraldSchafer  any update on this? 
Thanks a lot

qadmus, 2021-03-11:
I'll work on GM rate, torque values. Will take a week.

jyoung8607, 2021-03-11:
> * EPS torque (the torque added by the EPS on the steering wheel) in NM
>
> VOLKSWAGEN GOLF

I don't think I have this information to give you, at least not from our current bus position. Will it become important? I *might* be able to get it from a subset of cars with custom harnesses (gateway integrated), or a proxy value like motor current.

HaraldSchafer, 2021-03-12:
It could be important in the future. Since we control on torque it would be good to know how the car responds to torque. If we can get the EPS torque when OP is not engaged, then we can "calibrate" controls to have better feed forward specific to every car.

However, if it needs a custom harness, will probably have to find a different approach for those cars.

qadmus, 2021-03-25:
Heads up on side effects of changing units of human torque (CS.steeringTorque) to Nm.
apply_std_steer_torque_limits() mixes the steering command units with the driver torque units.

sshane, 2021-06-23:
I'm going through each message's bits and comparing it to the sign of the of the commanded torque and so far I haven't found any matching bits on bus 0, 1, or 2 for the few Hondas and Subarus I've looked at so far (HONDA ODYSSEY 2018 EX-L, HONDA ACCORD 2018, HONDA INSIGHT 2019, SUBARU ASCENT LIMITED 2019, SUBARU IMPREZA LIMITED 2019).

Tested the method on the Toyota Corolla and it found the correct sign bit. Is it possible that these cars don't have EPS torque exposed?

sshane, 2021-06-29:
I went through all the bits in a few routes for the Hyundais (since they all use the same DBC) and tried to find the sign bit for EPS torque from the sign of the commanded torque at that time (filtering for engaged samples). But the closest bits were mismatched 30% of the time (mismatch is the bit being 1 when the torque cmd was positive, etc.).

```
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=2
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=3
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=4
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=5
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=6
hex(msg)='0x156', bit_mismatches=9757 of 32127 (30.37%), byt_idx=2, bit_idx=7
Searched bus: 0
```

And if you assume the EPS signal is flipped, it gave similar results:

```
hex(msg)='0x335', bit_mismatches=764 of 2178 (35.08%), byt_idx=2, bit_idx=6
hex(msg)='0x374', bit_mismatches=761 of 2195 (34.67%), byt_idx=3, bit_idx=3
hex(msg)='0x18f', bit_mismatches=6635 of 21916 (30.27%), byt_idx=0, bit_idx=7
```

While running the same script on a 2017 Toyota Corolla with a known EPS signal, it produced the correct output finding the start of the EPS torque signal from cmd torque:

```
hex(msg)='0x260', bit_mismatches=316 of 3952 (8.0%), byt_idx=5, bit_idx=1
hex(msg)='0x260', bit_mismatches=20 of 3952 (0.51%), byt_idx=5, bit_idx=5
hex(msg)='0x260', bit_mismatches=2 of 3952 (0.05%), byt_idx=5, bit_idx=3
hex(msg)='0x260', bit_mismatches=1 of 3952 (0.03%), byt_idx=5, bit_idx=7
Searched bus: 0
```

So it's possible that the true EPS torque command is on another bus, or not available here.

sshane, 2021-06-29:
For the Hyundais, this was the closest I got for the angle rate (sign was calculated from the change in angle from carState):

```
hex(msg)='0x5b0', bit_mismatches=12 of 29 (41.38%), byt_idx=0, bit_idx=0
hex(msg)='0x386', bit_mismatches=595 of 1448 (41.09%), byt_idx=3, bit_idx=5
hex(msg)='0x220', bit_mismatches=1169 of 2881 (40.58%), byt_idx=2, bit_idx=7
hex(msg)='0x386', bit_mismatches=583 of 1448 (40.26%), byt_idx=7, bit_idx=5
hex(msg)='0x386', bit_mismatches=579 of 1448 (39.99%), byt_idx=1, bit_idx=4
hex(msg)='0x251', bit_mismatches=1115 of 2887 (38.62%), byt_idx=7, bit_idx=6
```

Where again it could find the signal for the Corolla and other known angle rate vehicles.

Jon8RFC, 2021-07-07:
How can I help with the Hyundai Sonata 2019 (2018 here)?  I'm not a software developer, but I'm happy to run a debug fork or be on the phone with one of you having ssh access and do a "turn the wheel all the way to the right" kind of thing.

adeebshihadeh, 2022-10-13:
No longer necessary

closed 2022-10-13 by adeebshihadeh