commaai/openpilot #20263: Offroad UI enhancements

adeebshihadeh, 2021-03-07:
- [ ] Allow editing tethering password while enabled
- [ ] Show username for current SSH keys
- [ ] Refresh developer page periodically
- [ ] Don't shift drive stats down when alert/update button is shown
- [ ] Click outside of confirmation dialog to dismiss
- [ ] Close confirmation dialogs after settings timeout

BrevanJ04, 2021-03-08:
- [ ] WPA2-EAP support
- [ ] Remove saved wifi networks
- [ ] UI change to denote the difference between connecting over wifi and ethernet

doktorsleepelss, 2021-03-09:
* [ ]  View mount placement degrees like on C2

jyoung8607, 2021-03-10:
- [ ] Optionally mask Wifi password during entry

speedking456, 2021-03-13:
- [ ] Reset SSH keys to default

michaelhonan, 2021-03-15:
- [ ] comma prime upgrade widget overlaps the home statistics view.

grekiki, 2021-03-15:
> * [ ]  comma prime upgrade widget overlaps the home statistics view.
I think that's only true on PC because it uses the narrower screen(https://github.com/commaai/openpilot/blob/fb5873bb6d00b4f625e9ee0cd1767fe8805b2476/selfdrive/ui/qt/qt_window.hpp#L16)

closed 2021-04-24 by adeebshihadeh