commaai/openpilot #20269: HEVC log file not created correctly during CARLA simulation.

davidtena98, 2021-03-07:
**Describe the bug**
 When I run a `simulation` on CARLA, I get the log video files in hevc.mkv format. When I try to open the video with VLC, only a quick visualization of the video is shown (less than half a second). Then the video stops.
**How to reproduce or log data**
It happens every time I run a simulation and a video/log files are created. To do this I followed the installation instructions.

**Expected behavior**
A creation of a correct HEVC file.

**Additional context**
I tried to find the problem by checking the VLC log file. It says "mkv error: cannot use the segment". Then I tried to extract the frames of the video using ffmpeg. I only could get 2 frames. However, the video file that I used is +40 Mb, which corresponds to around 20 seconds of simulation.

The bz2 log file seems to be ok.

Operating system: Ubuntu 20.04

pd0wm, 2021-03-08:
HVECs are currently encoded using OMX which I don't think has a PC version. So on the computer we store the logs as raw.

If you want to implement this you need another implementation of `VideoEoncoder` in `selfdrive/loggerd` powered by something like ffmpeg and use that on the PC.

godoppl, 2021-03-28:
Would implementing [libde265](https://github.com/strukturag/libde265) be a viable solution?

pd0wm, 2021-03-29:
Since it's for development only I don't attach too much value to the exact backend. As long as it's not too much code and nicely abstracted in a VideoEncoder class it's fine.

adeebshihadeh, 2021-06-18:
Closing this as there's nothing really to track here. This is more of a nice to have.

closed 2021-06-18 by adeebshihadeh