commaai/openpilot #20293: Open pilot 082 lateral control issues

josequirozramos, 2021-03-10:
**Describe the bug**
Open pilot 082 can’t handle curves like 081 used to do (lateral control) It’s actually dangerous now and not fun to drive. I only experienced this issue after upgrading and driving on roads I know my comma 2 op 081 could handle with ease. What happened?

**How to reproduce or log data**
Just drive with me, I can take my comma in if you want I’m in San Diego 

**Expected behavior**
To be the same or better not worse than 081 at taking curves. Seems sluggish often time invading the other lane

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: ab47547bd899515c
 - Route: all of them
 - Timestamp: all of them
 - Version: 0.8.2-release
 - Car make/model: Toyota Prius 2017
**Additional context**

I also get an ocasional GPS error

josequirozramos, 2021-03-10:
FYI I have done a factory reset already and done the calibration but it's still exactly the same.

pd0wm, 2021-03-10:
The bug report is too vague for us to really look into it. Can you give us two routes of the same road with 0.8.1 vs 0.8.2?

josequirozramos, 2021-03-11:
OK I will figure out how to install the older version first￼ 0.8.1 drive and then install 0.8.2 drive So we can rule out everything else I just have the software be different￼. Do you have a recommended way to install ￼older versions?

pd0wm, 2021-03-11:
All release are tagged: https://github.com/commaai/openpilot/tags

josequirozramos, 2021-03-12:
tagged? What does that mean, I am just trying to downgrade to older version to help fix the problem.

michaelhonan, 2021-03-15:
@josequirozramos If you want comma to look into your issue, you need to find an example in footage (https://my.comma.ai), then click "Copy Current Segment", and update your issue with the segment text it copies. Otherwise comma is going to waste precious time hunting for issues.

josequirozramos, 2021-03-15:
Thanks, this is where 0.8.2 goes outside of the lane at 50mph: ab47547bd899515c|2021-03-12--20-55-55--10
0.7.1 does do that at 50mph.

pd0wm, 2021-03-16:
Do you also have the logs for an older release that could take that turn. Were you running 0.7.1 before this? That version is almost over a year old. Would be better if you have a more recent release that still works.

josequirozramos, 2021-03-18:
Fair enough @pd0wm maybe on the weekend I can drive the same segment with 8.1 at the same speed and time. Yes maybe not call it a bug, 0.8.2 is growing on me.

josequirozramos, 2021-03-26:
Sorry @pd0wm I have been too busy have not had time to do this.

openpilot12, 2021-04-05:
I have a Subaru Ascent 2019 and have noticed that the 8.2 update causes the car to favor the left especially on turns. There have been several occasions where it would go onto the center divider line.

michaelhonan, 2021-04-06:
@openpilot12 0.8.3 has been released. Please upgrade, and try that. It has an updated model which has a much better lateral experience.

pd0wm, 2021-04-06:
Going to close this issue. Feel free to make a new one if you have issues with 0.8.3

closed 2021-04-06 by pd0wm