commaai/openpilot #20299: 2021 Toyota Highlander "Hold steering wheel" keeps popping-up on dash.

GeorgeM320, 2021-03-11:
Open pilot is running and working fine. I am still getting the hold steering wheel every 15 seconds or so. The car is a 2021 Toyota Highlander XSE

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

vanillagorillaa, 2021-03-11:
Please edit your issue and provide all of the Device information. There's nothing for us to look at otherwise.

GeorgeM320, 2021-03-12:
Sorry new to this.
Comma2 
Dongle a6c393a6d83cdb97
serial  287e29ee
version v0.8.2
2021 Toyota Highlander XSE
I was under the impression from videos I watched the "hold Steering wheel" warning on dash is disabled with the comma2. 
If the unit is enabled and I let go of the wheel, 14 seconds later the "LTA Hold Steering Wheel" warning appears on the dash. If I continue to not touch the wheel the cruise/auto pilot disengages.

vanillagorillaa, 2021-03-12:
The problem is how your device is mounted. If you review the driver footage on explorer (my.comma.ai find a drive and select "Driver Video to the top right of the video on the page) you can see that only half of your face is shown. You will want to remount your device in a way that the device can fully see your face. 

You can use the "Driver Camera View" (found in settings -> Device Settings) while the car is off and make sure that your full face is seen in the unshaded portion of the preview. 

For more help I would suggest to checkout our community discord, discord.comma.ai, where users there can help you out.

closed 2021-03-12 by vanillagorillaa