commaai/openpilot #20314: Speed Too High alert on german roads does not make sense

razem-io, 2021-03-11:
**Describe the bug**

Since version 0.8.2 there is an alert shown when driving ~ 90 mph. It shows "Speed Too High -> Slow down to resume operation" but everything is fine. C2 works as expect. Still the alert persists.

At least the stock version of the nissan leaf allows ProPilot speeds until ~90mph.

**How to reproduce or log data**

Drive above 88 mph.

**Expected behavior**

Do not display this warning until 95 mph. This would allow german drivers to use their C2 in peace at a slow pace.

**Device/Version information:**
 - Device: comma two
 - Version: release2 / 0.8.2
 - Car make/model: Nissan Leaf 2.ZERO 2018

HaraldSchafer, 2021-03-11:
We discovered the model can make strange/unexpected predictions at high speeds. So we lowered the max speed. If we can get the model to perform better at higher speeds, we can increase the max speed again.

If you want this too happen sooner, making sure your device is connected to the internet as much as possible. We don't have much autobahn data in the dataset, the more we get the quicker the model will do well there and we can up the speed again.

closed 2021-03-11 by HaraldSchafer

razem-io, 2021-03-12:
Thank you for the quick response. Can we leave this issue open until it is resolved?

Also the car does not disengage at high speeds. It shows an alert that it is no longer active, which is false. Wouldn't it make more sense to disengage, as it says in the warning or to alter / create a new alert which just displays a warning that it might behave unexpectedly? I am happy to make a PR if this would be a welcome addition to OP.

I upload 50 km of autobahn drive usually once a day. Currently blocked a little by Corona.

razem-io, 2021-03-12:
@HaraldSchafer sorry, I missed https://github.com/commaai/openpilot/pull/20322/files. Thank you very much! I am fine with this being closed then.

HaraldSchafer, 2021-03-12:
No problem!