commaai/openpilot #20323: Driver model lagging

oboist24, 2021-03-11:
**Open pilot disengages after 5-10 seconds due to driving model lagging**

 - Device: [comma two]
 - Dongle ID: comma two 1798c611fdd4a883 
 - Route: 3-10-21 and 3-11-21
 - Timestamp: [Within seconds of engaging openpilot. 
 - Version: [0.8.2]
 - Car make/model:  Toyota Prius 2020]

pd0wm, 2021-03-12:
What are the exact route names? (see https://my.comma.ai/useradmin/?onebox=1798c611fdd4a883)

oboist24, 2021-03-12:
Date:                       Time:
2021-11-03            05:41-05:45
2021-10-03            16:18-16:48
2021-10-03             16:14-16:17
2021-10-03             16:07-16:13
2021-10-03              15:11-16:02
2021-10-03            15:00-15:09
2021-10-03            05:40-07:30

pd0wm, 2021-03-12:
Looks like the road camera is outputting sequential frames with the same frame id, which makes modeld think it's lagging.

oboist24, 2021-03-12:
How can it be fixed?

ZwX1616, 2021-03-12:
Hi. Did you only experience these since version 0.8.2? Would be helpful if you let us know

oboist24, 2021-03-12:
Yes this is recent and unfixable on my end. I have rebooted daily to no
avail.

ZwX1616, 2021-03-16:
Looks like it's device related. Support should reach out soon. Thanks.

closed 2021-03-16 by ZwX1616