commaai/openpilot #20333: Longitudinal control slow to react

NonDonRandy, 2021-03-12:
**Describe the bug**

After getting 0.8.2, I have noticed that longitudinal control is extremely slow to react, and will sometimes slow down (due to an obstacle/vehicle) but the. Never get back up to speed.

**How to reproduce or log data**

OP running and engaged, follow vehicle at a higher set point than they are moving. Lead vehicle moves (allowing space to speed up). OP will maintain old speed.

**Expected behavior**

Speed up when gap appears

**Device/Version information (please complete the following information):**
 - Device: C2
 - Dongle ID: c49de5fce000d77
 - Route: https://my.comma.ai/useradmin/?onebox=c4f9de5fce000d77%7C2020-12-26--10-42-43
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: 0.8.2
 - Car make/model: 2021 Prius

**Additional context**

https://youtu.be/LCZxHrFczQ0 

Video of issue

adeebshihadeh, 2021-03-13:
The given route is from December 2020 running openpilot v0.7.9. Do you have a specific route and segment for this issue on 0.8.2?

NonDonRandy, 2021-03-14:
Crap, that was the file I preserved. Can I give you access to my device?

It seems to only do it during the day. Here is the drive it did it on 

https://my.comma.ai/useradmin/?onebox=c4f9de5fce000d77%7C2021-03-12--05-25-43

Art

adeebshihadeh, 2021-06-17:
We have access to all the logs. Is this still an issue? If so, we'll need a route where this happened. The route will need to be fully uploaded, so make sure to enable the "Upload Raw Logs" toggle.

closed 2021-09-04 by adeebshihadeh