commaai/openpilot #20367: cereal/messaging/socketmaster.cc:34: undefined reference to `AlignedBuffer::~AlignedBuffer()'

peterdegn, 2021-03-16:
**Describe the bug**
Operating system: Ubuntu 20.04
i get a error in ui.py there say "No module named cereal.messaging.messaging_pyx 
but i have just install it from the cereal folder no errors for the install of that

then i run : scons -j4 from the openpilot folder 

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o selfdrive/boardd/boardd -Wl,-rpath=/home/ubuntu/openpilot/phonelibs/snpe/x86_64-linux-clang -Wl,-rpath=/home/ubuntu/openpilot/cereal -Wl,-rpath=/home/ubuntu/openpilot/selfdrive/common selfdrive/boardd/boardd.o selfdrive/boardd/panda.o selfdrive/boardd/pigeon.o -Lphonelibs/snpe/x86_64-linux-clang -Lphonelibs/libyuv/x64/lib -Lcereal -Lselfdrive/common -L/usr/lib -L/usr/local/lib -Lcereal -Lphonelibs -Lopendbc/can -Lselfdrive/boardd -Lselfdrive/common -lusb-1.0 selfdrive/common/libcommon.a -ljson11 cereal/libcereal.a cereal/libmessaging.a -lpthread -lzmq -lcapnp -lkj
ar rc selfdrive/boardd/libcan_list_to_can_capnp.a selfdrive/boardd/can_list_to_can_capnp.o
ranlib selfdrive/boardd/libcan_list_to_can_capnp.a
clang++ -o selfdrive/camerad/cameras/camera_common.o -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -I. -Iselfdrive -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iselfdrive/boardd -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 selfdrive/camerad/cameras/camera_common.cc
clang++ -o selfdrive/camerad/transforms/rgb_to_yuv.o -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -I. -Iselfdrive -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iselfdrive/boardd -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 selfdrive/camerad/transforms/rgb_to_yuv.cc
clang++ -o selfdrive/boardd/boardd_api_impl.so -pthread -shared -Wl,-rpath=/home/ubuntu/openpilot/phonelibs/snpe/x86_64-linux-clang -Wl,-rpath=/home/ubuntu/openpilot/cereal -Wl,-rpath=/home/ubuntu/openpilot/selfdrive/common selfdrive/boardd/boardd_api_impl.o -Lphonelibs/snpe/x86_64-linux-clang -Lphonelibs/libyuv/x64/lib -Lcereal -Lselfdrive/common -L/usr/lib -L/usr/local/lib -Lcereal -Lphonelibs -Lopendbc/can -Lselfdrive/boardd -Lselfdrive/common -lcan_list_to_can_capnp -lcapnp -lkj
cereal/libmessaging.a(socketmaster.os): In function `SubMaster::SubMessage::~SubMessage()':
/home/ubuntu/openpilot/cereal/messaging/socketmaster.cc:34: undefined reference to `AlignedBuffer::~AlignedBuffer()'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/boardd/boardd] Error 1
clang++ -o selfdrive/camerad/imgproc/utils.o -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -I. -Iselfdrive -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iselfdrive/boardd -Iselfdrive/common -Iselfdrive/camerad -Iselfdrive/camerad/include -Iselfdrive/loggerd/include -Iselfdrive/modeld -Iselfdrive/sensord -Iselfdrive/ui -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 selfdrive/camerad/imgproc/utils.cc
scons: building terminated because of errors.

"
home/ubuntu/openpilot/cereal/messaging/socketmaster.cc:34: undefined reference to `AlignedBuffer::~AlignedBuffer()'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [selfdrive/boardd/boardd] Error 1
"
i not sure where to move on from here.

peterdegn, 2021-03-16:
i fix it 
./configure
make -j6 check
sudo make install

closed 2021-03-16 by peterdegn

awd2203, 2021-03-23:
There is no configure file in ~/openpilot 
how can you make & make install?

AIasd, 2021-06-02:
Hi @peterdegn  I encountered the same error. Could you let me know what did you do to fix the error? I did not find the configure file you mentioned. Thanks a lot!