commaai/openpilot #20375: KIA ceed 2019 dashboard warning

ZanZaD13, 2021-03-16:
FCA (Forward collision assitence) Dashboard warning upon engage.

When engaging OP FCA warning comes up on dashboard and stays until disengage.
Warning can be delayd 5-10 seconds when both lanelines are visible.

Issue has ocurred since either 0.7.4 or 0.7.5 since this was not happening on 0.7.3

**Device/Version information (please complete the following information):**
 - Device: Comma two
 - Dongle ID: e0e98335f3ebc58f
 - Route: e0e98335f3ebc58f|2021-03-14--16-54-08
 - Timestamp: Upon every engage. 17:59
 - Version: master 0.8.3
 - Car make/model: Kia ceed 2019

VirtuallyChris, 2021-03-17:
Try adding your car to the 20hz LFA MFA message area. A similar PR is open here https://github.com/commaai/openpilot/pull/20320

If that fixes it, submit a PR and we'll merge it.

ZanZaD13, 2021-03-17:
Car is allready there, does not change the problem.

adeebshihadeh, 2021-06-19:
Is this still an issue?

ZanZaD13, 2021-06-19:
Yes this is still happening in 0.8.5.

sshane, 2022-06-11:
Are you still experiencing this issue?

ZanZaD13, 2022-06-16:
> Are you still experiencing this issue?

Yes this is still an issue on 0.8.13.1 Release2

Why isn't it auto updating to 0.8.14?

sshane, 2022-07-21:
@ZanZaD13 please try this branch if you're still using a comma two: https://github.com/commaai/openpilot/commits/fix-lfa-warn-c2

ZanZaD13, 2022-07-21:
I will try this as soon as i can, I'm still on C2.

ZanZaD13, 2022-07-23:
> @ZanZaD13 if you're still using a comma two, please try this branch and let me know how it goes: https://installer.comma.ai/commaai/fix-lfa-warn-c2
> 
> If you're using a comma three, try this branch: https://installer.comma.ai/commaai/fix-lfa-warn

I've only had time for a 5 minute drive, weird result, seemed like it was working on first engage, (op drove about 20 seconds without FCA warning) I had to make a turn, then on next engage i got FCA warning again..

sshane, 2022-08-01:
Hi @ZanZaD13, please try that branch again, either by using SSH and `git pull`, or re-install that branch entirely. Let me know with a route if it works!

ZanZaD13, 2022-08-04:
> Hi @ZanZaD13, please try that branch again, either by using SSH and `git pull`, or re-install that branch entirely. Let me know with a route if it works!

It doesn't work, if anything the error occurs instantly upon OP engage.

sshane, 2022-08-05:
Pushed another update, let me know how it goes

ZanZaD13, 2022-08-07:
> Pushed another update, let me know how it goes

FCA warning upon car startup (shows entire ride)
OP not working ( Steering currently unavailable ) when trying to engage.

sshane, 2023-08-07:
Can you send a route of this happening recently?

adeebshihadeh, 2023-09-02:
Closing due to inactivity. Post a route if it's still an issue and we'll reopen.

closed 2023-09-02 by adeebshihadeh