commaai/openpilot #20383: webcam on PC not work

lixit, 2021-03-17:
**Describe the bug**

no road image showed in the UI

I change the file `selfdrive/camerad/cameras/camera_webcam.cc`
to ` cv::VideoCapture cap_road(0); // road`
to use my PC webcam. And compile. But there is no road image in the UI

**How to reproduce or log data**

follow the steps on 
[https://github.com/commaai/openpilot/tree/master/tools/webcam](https://github.com/commaai/openpilot/tree/master/tools/webcam)

**Expected behavior**

With only a PC, and a webcam. Can it run?

**Additional context**

My PC has only 1 camera. I didn't connect panda. And the UI shows NO PANDA.
Is the panda and the second camera required?

Operating system: [e.g. Ubuntu 20.04]

adeebshihadeh, 2021-03-17:
A panda is required to go onroad, which is why you don't see the camera feed. What are you trying to use the webcam stuff for?

lixit, 2021-03-17:
It will be cool if it can run in my PC with nothing else. I want to test it but I don't have the panda hardware. I want to take my PC in a car and see the UI actions on the road.

adeebshihadeh, 2021-03-17:
Closing since the webcam is working as intended, but you can easily achieve what you want with some modifications. I'd recommend just getting the relevant daemons running (UI, modeld, camerad).

closed 2021-03-17 by adeebshihadeh