commaai/openpilot #20397: 2017 Subaru Impreza ping pong in lane

Jahmangy, 2021-03-18:
My 2017 Subaru Impreza does not drive straight between the lane lines to a point that it’s not useable. Help!!

pd0wm, 2021-03-18:
Please fill out the bug report template with dongle id and drive(s).

Jahmangy, 2021-03-18:
Hi Willemstad,
Where would I find the report template? And what’s the dongle Id?

Maurice 
Best Regards

pd0wm, 2021-03-19:
When you create a new issue on github it will populate with a template. Basically we want this info:

Device: [e.g. EON/EON Gold/comma two]
Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
Car make/model: [e.g. Toyota Prius 2016]

Jahmangy, 2021-03-19:
Device/Version information (please complete the following information):

Device: Comma2
Dongle ID: 0bad2474c0a9a035
Route: all but most recent 69
Timestamp: all 
Version: v0.8.2-release
Car make/model: 2017 Subaru Impreza with EyeSight
Additional context
The path mapping still showed the car is between to lane lines even when it crosses outside of the lane

RCoaster, 2021-03-27:
Steer Actuator Delay is too high- should be lowered by .2 to reflect the new MPC

Jahmangy, 2021-03-27:
Thanks. However, how do I go about making this fix? 

Maurice 
Best Regards

dbzx6r, 2021-03-28:
You will need to ssh and cd into /data/openpilot/selfdrive/car/subaru and use 'nano interface.py' to modify the value from 0.4 to 0.2. I'm in the same boat as you with pingponging, i've made this change and I'm going to test it out

dbzx6r, 2021-03-28:
i've made the change, and it is significantly better, but there's still some room for improvement.

in the video below specifically watch around the 1:15 mark, you can see it still gets pretty close to the lane line, and there is still pingponging BUT it will maintain the lane in this short drive.

[https://www.youtube.com/watch?v=n4QwyiiONUo](url)

do you have any suggestions on the next value to adjust?

michaelhonan, 2021-04-26:
Can confirm (as I have an 18 Impreza, which is the exact same car), this is not an OP bug. It's related to tuning.

Likely worth closing this as 0.8.3 corrected a lot of the lateral issues seen in 0.8.2. Rest can be tuned out.

Jahmangy, 2021-04-26:
OK. I'm still on 0.8.2 and the device is not giving me the option to do the OTA update to 0.8.3. I tried powering down and resting the device but it's still on 0.8.2.

michaelhonan, 2021-04-27:
@Jahmangy if you're not on stock, you'll need to ask your fork maintainer. Likely either "fk da police" or "mlp" in the Subaru channel on Discord.

Jahmangy, 2021-04-27:
It's on stock. Everything is stock. I'm just trying to get it to work after my first experience with my initial drive.

michaelhonan, 2021-04-27:
Fair enough! Well, I'd connect to Wifi if you haven't already, and just let it sit for 5 minutes. It should prompt you.

If all else fails, just head to Settings -> Device -> Uninstall openpilot, then follow the [instructions to install openpilot](https://github.com/commaai/openpilot/wiki/Installing-openpilot) again.

adeebshihadeh, 2021-05-04:
Closing this since it sounds like a tuning issue and lateral performance on subarus should be improved with recent versions.

closed 2021-05-04 by adeebshihadeh