commaai/openpilot #20409: Permanent green alert for dashcam mode

HaraldSchafer, 2021-03-19:
It isn't immediately clear whether the device is running in dashcam mode. It should be a permanent green message, or similar.

pd0wm, 2021-03-22:
Guess we should only show it permanently if the dashcam mode is an unexpected situation (i.e. not the dashcam branch or not openpilot disabled using the toggle).

gregjhogan, 2021-03-23:
How about a different border color for dashcam mode, too?  I have switched to dashcam mode to test something with the stock system, forgot to switch it back, and then missed the short alert at the beginning the next time I drove.

michaelhonan, 2021-03-24:
Could the steering wheel icon change to a camera icon? Maybe that would also be clearer.

adeebshihadeh, 2021-03-26:
I think the different border color would be more confusing and any other UI changes should get some more thought. Added a small persistent alert in https://github.com/commaai/openpilot/pull/20483. The normal alert will show first explaining the reason for dashcam then you'll get a small, one-line alert for the rest of the drive.

closed 2021-03-26 by adeebshihadeh

gregjhogan, 2021-03-26:
yeah different border color was only helpful if we didn't show the alert permanently when dashcam toggle is on