commaai/openpilot #20419: screen way too dark while driving in bright sunlight

gregjhogan, 2021-03-20:
**Describe the bug**

road image was really dark and the only thing I could see was the overlay while driving

**How to reproduce or log data**

drive in bight sunlight

**Expected behavior**

road camera view much brighter in similar conditions

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 1ad763dd22ef1a0e
 - Route: 1ad763dd22ef1a0e|2021-03-20--13-58-28
 - Timestamp: anytime during this drive, and several other surrounding routes the same day
 - Version: kl-model
 - Car make/model: Hyundai Palisade 2020

**Additional context**

gregjhogan, 2021-03-22:
forced brightness to always be the max and it didn't help, the overlays/sidebar are easy to see but the road image is still really dark while driving on a bright day

VirtuallyChris, 2021-03-22:
I noticed this. I think some brightness curve on the camera feed would be ideal when brightness is close to max. The screen really washes out those darker colors!

spektor56, 2021-03-27:
I have issues with brightness as well, I wonder if it's being caused by this?
#20127

cydia2020, 2021-04-01:
A headlight controlled display brightness would help with this - only dim the screen if the headlights are turned on, otherwise stay at full brightness, https://github.com/commaai/opendbc/pull/373

byxorna, 2021-04-27:
I also have this on a brand new c2 - the video feel has a desaturated blue tint to it, and as previously mentioned does not seem to impact the chrome, only the video feed.

adeebshihadeh, 2021-04-28:
Should be improved with 2c0b0f927fc517f728b6f35feded278e4f8152ce.

closed 2021-04-28 by adeebshihadeh