commaai/openpilot #20440: Dashcam Mode Car Unrecognized [2021 Toyota Rav4 XLE Premium]

bli1, 2021-03-23:
**Describe the bug**

unable to use open pilot

1. The car cannot be recognized by OP. 
2. Tried to follow fingerprint WIKI. BUT carFW section is empty and shown as "carFW = []"
3. Tried to modified values.py under directory (selfdrive/car/toyota/values.py) as described in https://github.com/commaai/openpilot/commit/d3c7c5f62a4f5e3d04fa492dd80aa8ebc9e2d0dc .
BUT, the devices still cannot recognize the car. 

**Expected behavior**

expecting comma2 recognize the car

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [f7837a0c8f262656]
 - Route: [f7837a0c8f262656|2021-03-22--16-52-43]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [0.8.2]
 - Car make/model: [2021 Toyota Rav4 XLE Premium]

vanillagorillaa, 2021-03-23:
Have you ensured that your comma power and ethernet cables is properly connected to the OBDII port and the harness box at the camera?

bli1, 2021-03-23:
> Have you ensured that your comma power and ethernet cables is properly connected to the OBDII port and the harness box at the camera?

Yes. Otherwise, it would show No Panda.

bli1, 2021-03-23:
> Have you ensured that your comma power and ethernet cables is properly connected to the OBDII port and the harness box at the camera?

No panda will not appear if I disconnect the ethernet cable. The car will popup errors if I disconnect the 10 or 12 pin connector. I guess I don't have a connection problem at the backside of the mirror. But I may have issue with the ethernet to OBDII interface. I just reconnected everything and the issue didn't disappear.

vanillagorillaa, 2021-03-24:
Looks like you opened another issue for this. Will close this one. #20452

closed 2021-03-24 by vanillagorillaa