commaai/openpilot #20445: Freeway lane marking tracking

2021fb, 2021-03-23:
**Describe the bug**
Good evening. I'm reporting a comma lane marking tracking problem. Before version .82
My comma would track freeway lane marking and center vehicle with minimal left or right corrections. Now. The tracking markers are (inside) the actual freeway marking. Causing the vehicle to drift back and forth inside the actual freeway marking. The vehicle is either to far left or right and occasionally making contact with the roadway bumps and on gentle curves. I manually hold the wheel to prevent oversteer.
Any help to correct this problem would be greatly appreciated.
Toyota Rav4 2018 Hybrid 
Fred Batin. 

![20210321_175904](https://user-
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

images.githubusercontent.com/81249926/112173926-3df7f800-8bb3-11eb-984b-3249645103b5.jpg)
[image]

pd0wm, 2021-03-23:
Looks like a calibration issue.

What is your dongle id? Please fill out the standard template:
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

adeebshihadeh, 2021-03-31:
Closing due to inactivity. Please re-open after filling out the template.

closed 2021-03-31 by adeebshihadeh