commaai/openpilot #20452: CarFW is empty.

bli1, 2021-03-24:
**Describe the bug**
I recently received my OP and tried to follow the fingerprinting WIKI. 
In the carParam section, the carFW is empty. 

In the unrecognized car thread, I was suggested to check the connection between cables. I reconnected everything and replaced ethernet cable and usb-c cable. It's not helping. 

I assume "carFW" is a different problem other than unrecognized car issue, thus I create this thread. 
  
**How to reproduce or log data**
1. connect comma and all other things. 
2. drive for a while. 
3. check the recent upload data
4. find carFW
5. carFW is empty. shown as carFW=[] !!! ISSUE HERE. 

**Expected behavior**
Expecting to find carFW to finish the fingerprinting WIKI instruction. 

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Version: [0.8.2 / 0.8.3]
 - Car make/model: [2021 rav4 ]

adeebshihadeh, 2021-03-24:
What's your dongle ID?

bli1, 2021-03-24:
> What's your dongle ID?

f7837a0c8f262656

bli1, 2021-03-24:
> What's your dongle ID?

Have you seen similar issues before?

bli1, 2021-03-24:
> What's your dongle ID?

It seems the device fails to communicate with the car. Do you consider it a hardware issue?

vanillagorillaa, 2021-03-26:
Please email support@comma.ai with your order number and the team can take care of it for you.

closed 2021-03-26 by vanillagorillaa