commaai/openpilot #20480: Latest master-ci update caused black screen of death

cferra, 2021-03-26:
**Describe the bug**

After latest update - comma 2 rebooted and it resulted in a black screen

**How to reproduce or log data**

Update to latest master ci build from device

**Expected behavior**

reboot and device to boot as normal

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 2876aae3c6df4380
 - Version: latest master ci build
 - Car make/model: 2019 Toyota CH-R

NikitaNekrasov, 2021-03-26:
Likewise on Hyundai Ioniq EV LTD. Going into recovery mode, reinstalling OpenPilot, switching to master and updating submodules works but clicking “Reboot and Update” when promoted on the UI will cause this endless black screen (1hr+) as well.

adeebshihadeh, 2021-03-26:
I can't reproduce this running master-ci. Master-ci was briefly missing a spinner, meaning you'd see a black screen while it was building openpilot. Is this still an issue?

cferra, 2021-03-26:
It just sat there on the black screen. Did not do anything - pressing the power button showed the power off and reboot dialog so system was fully booted

adeebshihadeh, 2021-03-26:
There was also an issue with this commit 3d48bd934d6c84a5352c2a1d56e6f93ed59e28c4. Can you check what commit you're on and update manually?

cferra, 2021-03-26:
I’m back on release2 atm

adeebshihadeh, 2021-03-26:
Then I think we can attribute this issue to 3d48bd934d6c84a5352c2a1d56e6f93ed59e28c4, which is fixed, unless @NikitaNekrasov can confirm this is still an issue.

NikitaNekrasov, 2021-03-26:
Did a git deinit —all -f, git checkout origin/master-ci, gif fetch, gif pull all from master originally and now master-ci seems to work. Have not been able to try the “Reboot and Update button on the UI since I haven’t been promoted for it yet.

adeebshihadeh, 2021-03-26:
Closing as this seems to be resolved. Please reopen if it happens again.

closed 2021-03-26 by adeebshihadeh