commaai/openpilot #20517: When the lead car creeped ahead, OP failed to stop

adnanlanewla, 2021-03-28:
**Describe the bug**
I was behind a lead car. OP stopped as expected for the lead but as soon the lead car creeped a head, it almost tried to crash in to the lead car.

I was driving on the stock OP. Usually OP never has a problem in stopping for a lead car. But this one time, OP stopped for the lead car and then as soon as the lead car creeped ahead, OP tried to accelerate abruptly and was about to crash in to the car. I had to manually take over and put the brakes on the car. If I were to not do anything this would have definitely resulted in a bad accident. 

**How to reproduce or log data**
I have only seen this bug happen once. So I can't reproduce it.

N/A

**Expected behavior**
When the lead car creeps ahead, OP shouldn't accelerate abruptly and should stop for the lead car.

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two] (Comma2)
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin] (d2ef5aa690e849d0)
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin] (d2ef5aa690e849d0|2021-03-23--21-57-27)
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)] (12 - 13 min)
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)] (release2)
 - Car make/model: [e.g. Toyota Prius 2016] (Toyota Rav4 2020)

**Additional context**
https://www.youtube.com/watch?v=UT53QdPZEyU&ab_channel=Adi
This is the video of stitch segment from 12 - 13 mins.  Notice the video from 0.30secs to 0.50secs for the behavior I experienced.

pd0wm, 2021-03-29:
Looks similar to https://github.com/commaai/openpilot/issues/2731. Openpilot was commanding quite a bit of deceleration, but the car/PCM was very slow to respond. We have to understand why this happens, and maybe add some alerts if we can detect this.

[image]

pd0wm, 2021-04-08:
Closing this issue since we already have another one to track this specific problem and no more feedback was provided.

adnanlanewla, 2021-04-08:
You can close this issue if we have another bug tracking the same issue. I have not seen this issue happen again since I first reported. However, when it did happen it's quite dangerous so I think it's something to keep an eye on.

closed 2021-04-08 by pd0wm