commaai/openpilot #20531: Hyundai Santa Fe 2020 CAN Error sinced 0.8.2, HKG_community branch works fine

Wykillin, 2021-03-30:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 21adf1570c3e619e
 - Route: 21adf1570c3e619e\|2021-03-28--23-11-29--0

 - Timestamp: any
 - Version: release2, master, and master-ci
 - Car make/model: Hyundai Santa Fe 2020

**Additional context**
I am getting a can error on the more recent release versions of openpilot. After checking all of the connections and replacing the wires, I tried  the HKG_community branch again. I'm no longer getting a can error. I just turned the car on and checked. So, I've narrowed it down to a software issue (probably the safety checks) for my car.

adeebshihadeh, 2021-04-13:
Is this still an issue? I checked `21adf1570c3e619e|2021-04-08--17-52-22` and couldn't reproduce the issue. If it's still an issue, can you post a fully-uploaded route from master/master-ci?

Wykillin, 2021-04-14:
This is still an issue. I just checked out master and the can issue still
persists. Check this?
https://my.comma.ai/useradmin/?onebox=21adf1570c3e619e%7C2021-04-13--20-22-55

Thanks

adeebshihadeh, 2021-04-14:
Fixed in 9ddb12a69e16a8043f8f7bb2d745a649d82b6a34. This wasn't caught by our tests because we don't have a good test route for the Santa Fe. If you give us a route that you're ok with making public, we can make sure this doesn't happen for your Santa Fe again. It doesn't have to be anything special, just a couple minutes long.

closed 2021-04-14 by adeebshihadeh

Wykillin, 2021-04-14:
Awesome, thank you so much. I'll go for a drive later today and then figure
out how to make the route public.