commaai/openpilot #20552: ACC issue on 2017 Civic Sedan (Nidec) .8.3 and above (KL model)

DorkSaber, 2021-04-01:
**Describe the bug**
Openpilot no longer sees moving or stopped cars consistently in its path, several near collisions in 2 days of driving on master channel .8.3 and .8.4. 2017 Honda Civic sedan (Nidec). Stock ACC system works as it should, but when C2 is hooked up and OP enabled, steering is perfect but it does not detect traffic in front either until very close (4 car lengths or less) or at all sometimes, at any speed.

**How to reproduce or log data**
Travel at any speed with OP engaged. When pulling up to another vehicle in front, even one moving just slightly slower than mine, there is no triangle, no ACC "car" icon on the gauge cluster, as if its not there. .8.2 and prior versions picked up everything, stopped cars, golf cars, ATV's, motorcycles, basically 100% of the time. .8.3 either detects very late (just a few car lengths behind) or doesn't detect at all, to the point where the stock AEB warning flashes warning of imminent collision. 

**Expected behavior**
Prior versions (.8.2 and prior) all detected cars/trucks/ATV's/motorcycles/etc, well in advanced when moving, and when stopped at least early enough for OP to stop the car when traveling less than 45 mph. 

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: bc014e9a661d4a5e
 - Route: no routes found in comma.ai explorer, not sure if this is another bug or if the routes got cleared when I switched back to the release channel
 - Timestamp: around 8:30pm eastern 3/31/2021 was a serious one, when a car slowed down to turn right in front of me on a perfectly straight road, regular event on a road I travel on several times a day and OP always catches it and slows down or stops until the vehicle in front is clear of the road. Or any trip from 3/31/2021, several examples of this in every trip.
 - Version: .8.3 and .8.4 master
 - Car make/model: 2017 Honda Civic sedan

**Additional context**

pd0wm, 2021-04-01:
This should be the explorer segment: https://my.comma.ai/107946debf4935ed/1617237163136/1617237197877

`107946debf4935ed|2021-03-31--20-25-03` end of segment 7

HaraldSchafer, 2021-04-01:
The failure in the segment you showed is a problem, but I'm not convinced it's a regression. Openpilot has often struggled with approaching slower leads at night. Closing this issue, feel free to open a new issue if you have an example that is more extreme.

closed 2021-04-01 by HaraldSchafer

DorkSaber, 2021-04-01:
I think that's a huge mistake. It is a major regression. This kind of
failure has never happened to me before, same routes same car, for about a
year now. It happened nearly 50% of the time in 2 days of .8.3. But that's
fine, I'm going to stick to .8.2 and staying there I just hope no one gets
hurt. If I can get my explorer working (still shows no drives for past 2
weeks) I will send you plenty of time stamps so you can see them all.

-Darek

pd0wm, 2021-04-01:
Looked more into other cases that were more clear cut and it looks to be a radar alignment issue.

DorkSaber, 2021-04-01:
Yes thank you again. Your explanation makes sense. I will get it fixed.
Thank you again!