commaai/openpilot #20554: Lanes falsely extrapolated due to lead car cut in and offramp exit

Jon8RFC, 2021-04-01:
**Describe the bug**

This is a pretty interesting scenario, unless you've seen it before.
Lane lines briefly broke on the right side for an offramp exit, while simultaneously a new lead car cut across from the left to the right to exit.  OpenPilot seems to have interpreted this as the lane curving to the right once the left side lane line was blocked from view.

**Expected behavior**

OpenPilot would not judge a cut-in vehicle to cancel out the lane lines previously identified because they're temporarily not in view due to the new lead car.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 6363a0e48f515f27
 - Route: 6363a0e48f515f27|2021-03-20--13-30-49--65
 - Timestamp: 14:36:48 steering override in explorer
 - Version: 0.8.3-4f89f42d-2021-03-20T08:58:33
 - Car make/model: Hyundai Sonata 2018

Jon8RFC, 2021-04-01:
#19586

HaraldSchafer, 2021-04-01:
This unfortunately isn't really an actionable issue. There are no issues with the ground truth that cause this. It's just the model being confused, will improve as the model gets smarter.

closed 2021-04-01 by HaraldSchafer

Jon8RFC, 2021-04-01:
Just wanted to clarify that it wasn't staying on the correct path. I just reacted quickly and forced it to not exit when it began to wrongly exit. Is that normal confusion of the model in that scenario, to exit?