commaai/openpilot #20556: Honda Odyssey 2021

tovialefkowitz, 2021-04-01:
I have a 2021 Honda Odyssey. And the comma doesn’t recognize the car if someone can help me

pd0wm, 2021-04-01:
Please fill out the bug report template:
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

vanillagorillaa, 2021-04-01:
The 2021 moved from Nidec to Bosch hardware and will require a car port. You can see this discussion regarding that here, https://github.com/commaai/openpilot/discussions/2349

You can talk to @csouers on our community discord, discord.comma.ai, and he should be able to get you set up with his branch.

vanillagorillaa, 2021-04-01:
See the discussion linked above and join our discord community. The users there can help get you setup.

closed 2021-04-01 by vanillagorillaa