commaai/openpilot #20559: Kia Niro EV 2020 - CAN Error

speedking456, 2021-04-01:
**Describe the bug**

I've been working with @freiadam to get their Niro EV working with OP. OP was working at one point, with an error on the dashboard, however after attempting to change branches to remove the error, it hasn't been working since, with different errors in different forks. HKG gives, waiting for controls, master/master-ci gives CAN error. My custom fork also gives CAN error on master.

**Log data**

59db3fd0b2784d27|2021-04-01--18-25-45

**Expected behavior**

Custom fork warning message, or the standard alert before driving.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 59db3fd0b2784d27
 - Route: 59db3fd0b2784d27|2021-04-01--18-25-45
 - Timestamp: N/A
 - Version: Log was taken on master-ci 0.8.4-387bae9a
 - Car make/model: Kia Niro EV 2020 - LHD EUR model

**Additional context**

New USB-C cable has been tried, resetting the comma two, along with: new fingerprint, hardcoded fingerprint, xx979xx HKG fork, and my custom fork.

Looking for troubleshooting suggestions or someone more experienced with log analysis.

speedking456, 2021-04-02:
After working with @freiadam some more, 0.8.1 does work, it does not give a CAN error. Route 59db3fd0b2784d27|2021-04-02--13-34-06 has the correct fingerprint and functioned correctly with zero errors (no dash error either).

adeebshihadeh, 2021-04-13:
Fixed in f308430c11e238e57dc4e1c659a96902ce6281cd

closed 2021-04-13 by adeebshihadeh