commaai/openpilot #20565: Make clear when too slow for lane changes

HaraldSchafer, 2021-04-02:
If you command a lane change under 45mph, nothing will happen. It it should be clear to the user that no lane change can be done because the speed is too low.

Perhaps instead of the message "Steer Left to Start Lane Change" that appears above 45mph something like **"Speed too low for lane change"**

VirtuallyChris, 2021-04-02:
Maybe we can make a way to show this with an icon or something similar. I feel like an alert would be distracting every time I turn my blinker on under 45mph.

adeebshihadeh, 2021-04-02:
I think a small, one-line alert with no sound should be fine.

VirtuallyChris, 2021-04-02:
True, 1 line alert would be okay

Winvoker, 2021-04-03:
I was looking to solve this , however I saw that, it is already implemented something like this. Isn't this part of code does what you were saying.

https://github.com/commaai/openpilot/blob/367aec54d1c4c5ed8d3f651d0b2f8b80fb0ac0e5/selfdrive/controls/lib/events.py#L179

advcu987, 2021-04-03:
I saw the unit tests fail for this PR, doesn't the `belowSpeed` field need to be added to the `LaneChangeState` enum in [openpilot/cereal/log.capnp](https://github.com/commaai/cereal/blob/e1741865f8117f27206b0225b4c48fa7432b830f/log.capnp#L803)?

```
  enum LaneChangeState {
    off @0;
    preLaneChange @1;
    laneChangeStarting @2;
    laneChangeFinishing @3;
    belowSpeed @4
;  }

```

Winvoker, 2021-04-03:
> I saw the unit tests fail for this PR, doesn't the `belowSpeed` field need to be added to the `LaneChangeState` enum in [...]

Thank you sir ! I was looking for this. I created a PR for cereal. https://github.com/commaai/cereal/pull/129

cydia2020, 2021-04-04:
Is this something that you would be interested in? I will make this easier to see if you give the go.
[image]

pd0wm, 2021-05-03:
@HaraldSchafer Do you think this is still an issue now that the lane change speed is lowered?

adeebshihadeh, 2021-05-05:
I think it's fine now with the 30mph limit.

HaraldSchafer, 2021-05-05:
I think it;s fine now!

closed 2021-05-05 by HaraldSchafer