commaai/openpilot #20566: Make clear whether openpilot is controlling longitudinal

HaraldSchafer, 2021-04-02:
It's not obvious whether your car has stock ACC or OP ACC. It should be easier to see. Perhaps change in the onroad ui, so it is always clear or in the device info in settings?

gregjhogan, 2021-04-02:
We have a steering icon, maybe add a cruise control icon (standard icon is a speedometer)
https://en.wikipedia.org/wiki/Cruise_control#/media/File:Cruise_Control.svg

doktorsleepelss, 2021-04-03:
Problem with just an icon is that people could think long isn't working instead of it not being supported.  Would still need something else. For unrelated reasons though, I would love an animated working speedometer icon like the steering wheel. Would be useful to have for replay reasons on explorer. 

Having it in device info would be good though.  Possibly along with other limitations like stop and go support. Problem with that is having people know and check it's there.  Add a settings overview to onboarding?

cydia2020, 2021-04-03:
Put a "No Longitudinal Control" alert at startup would do imo

pd0wm, 2021-04-05:
Or we remove the set speed completely if you don't have OP long?

dorkmo, 2021-04-05:
could put like "honda" under the cruise control logo

PR #20585 (2021-04-05): hide set speed for stock long
fixes #20566
closed 2021-04-05 by adeebshihadeh