commaai/openpilot #20575: Webcam tool doesn't work well.

zhuhj-tery, 2021-04-04:
**Describe the bug**
Hi,
We try to use the webcam tool of openpilot on a PC. We followed the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC to two web cameras and the black panda. These two web cameras were tested to be able to work well. The computer successfully recognized these three devices. However, it seems that the cameras didn't work since the ui showed "Camera Malfunction! Contact Support!". In the terminal, there are some error messages showing that the cameras are not alive:
[image]
[image]
[image]
Any comments or suggestions are appreciated.
Thanks!
**How to reproduce or log data**

We just run the command: PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py and started the car.

**Expected behavior**

The UI should show the road webcam's view.

**Additional context**
Operating system: Ubuntu 20.04
Web camera: Logitech C920 and C615

grekiki, 2021-04-04:
Does your PC have a good Nvidia GPU?

zhuhj-tery, 2021-04-05:
Thanks for the suggestions. I tried to run camerad separately and found out that the issue is caused by the connection problem between the camera and the PC. So, I rebuilt openpilot with the correct camera index and it works.
Thanks a lot.

closed 2021-04-05 by zhuhj-tery