commaai/openpilot #20581: Communication issue between processes when using webcam tool

zhuhj-tery, 2021-04-05:
**Describe the bug**

Hi,
We installed openpilot in the PC and tried to implement the webcam tool. After finishing the calibration process, we tried to realize the ACC function by pressing the corresponding button in the car, but the ui just showed "**Communication issue between processes**". Are there any possible solutions?
Thanks!

**How to reproduce or log data**

The toggle id is d8412d04a2ad517f.
The time stamp is 2021-04-04--20-21-35
.
**Expected behavior**

We expect that the ACC function can work well.

**Additional context**

Operating system: Ubuntu 20.04
Version: v0.8.4
Car: Toyota RAV4 2019

adeebshihadeh, 2021-04-05:
Multiple daemons crashed, including modeld, dmonitoringmodeld, and camerad. Aside from testing that it builds, the webcam port isn't really tested and isn't expected to be a plug and play experience.

zhuhj-tery, 2021-04-06:
Hi,

Thanks for your reply!

If I understand correctly, the webcam tool has not been fully tested so the code might not function normally and could not just plug and play?

Are there any possible reasons and solutions for the crash? It would be great if you can elaborate a little bit that how could we fix it.

We are trying to do some research experiments that "augment" some objects to the real-time data captured by camera and radar sensors (e.g., adding/rendering a vehicle that does not exist on the road to the video captured by the camera). Is that possible to be done by using the Comma Two devkit (or is that possible by using the webcam+PC solutions)? Any comments or suggestions will be deeply appreciated!

Thanks again!

adeebshihadeh, 2021-04-06:
The webcam code was in a working state when it was initially written, but the codebase has undergone several refactors since then. The crashes happen very early and didn't make it to the logs, so I can't check exactly what happened. Either a comma two or PC will work, but the PC route will be easier. I'd recommend seeking help on our GitHub Discussions and discord rather than issues.

closed 2021-04-06 by adeebshihadeh

LIHaoran-good, 2021-05-15:
Hi , I am trying to  run openpilot with webcam on PC now, I refer to this guideline that [Build openpilot for webcam,](https://github.com/commaai/openpilot/blob/master/tools/webcam/README.md) when entering the compile instruction "USE_WEBCAM=1 scons -j$(nproc)", report the error "scons: *** Import of non-existent variable "libkf" ",  I don't know what the issue is, do you know the problem is ? 
In addition, can I add your contact information to communicate with you.
Thanks very much.

Operating system: [Ubuntu 20.04]