commaai/openpilot #20582: Comma2 will not start on inginition

patcoghm, 2021-04-05:
**Describe the bug**

  Comma 2 should boot up and run with ignition activation.

 - Device: Comma 2
 - Dongle ID:  e4a16227cff292b5
 - Route: all
 - Timestamp: any
 - Version: 8.3
 - Car make/model: 2020 Toyota Prius Prime XLE DSU unconnected.

**Additional context** before new version 8.3 car ignition would start comma 2 if off

cydia2020, 2021-04-05:
Have you checked all your cables? Esp. the USB c cable

patcoghm, 2021-04-05:
Yes. Never remove Comma, always installed since new. But cables are again checked asthe normal ghost problem. Have you checked the comma 2 turned off and then restart by ignition yet.

https://github.com/commaai/openpilot/issues/20582#issuecomment-813194620

patcoghm, 2021-04-05:
I did finally get a bug report up on discord. Again power off comma 2 with the comma power button. Make sure car is totally off. Wait. Then turn car only on. Comma will not both as in car not driven in days and the comma standby hours (was 30 hours) is expired and comma is turned off. Panama is waiting for extinction and has always started comma to a boot startup.
Thanks

https://github.com/commaai/openpilot/issues/20582#issuecomment-813194620

adeebshihadeh, 2021-04-05:
I can't reproduce your issue. Besides turning on the C2 on ignition, is everything else working fine?

adeebshihadeh, 2021-04-07:
Closing due to inactivity. Feel free to reopen if you can verify the behavior is fine on previous releases, but I suspect you'll see the same thing if you revert to 0.8.2.

closed 2021-04-07 by adeebshihadeh

patcoghm, 2021-04-07:
You did not answer the question . If comma 2 is powered off and the car starts with ignition, will comma2 turn on and boot up the comma 2. It openes to the boot process and the comma is shown and it loads the software.

My car WILL start comma2 with ignition if comma2 is in standby waiting for the next run.  

THE QUESTION STILL IS , SHOULD comma2 boot comma2 from a power off state with ignition.

Closed #20582.

https://github.com/commaai/openpilot/issues/20582#event-4564327999