commaai/openpilot #20593: Car fw versions in useradmin don't have proper formatting

VirtuallyChris, 2021-04-06:
Before 0.8.3, carfw was shown in the same format as it is in the values.py file.

Since many people still use useradmin to fingerprint their cars, we should make the logs match what they would see in values.py to prevent PRs [like this](https://github.com/commaai/openpilot/pull/20583)

[image]

gregjhogan, 2021-04-06:
This was a temporary fix until capnproto 0.9.0 is released with https://github.com/capnproto/capnproto/pull/1143

VirtuallyChris, 2021-04-06:
Ah I see, sucks that there hasn't been a release in over a year :/

adeebshihadeh, 2021-05-04:
There's unlikely to be a new capnp release soon. @gregjhogan can we just build capnp from source in useradmin for now?

gregjhogan, 2021-05-06:
Tried using the bleeding edge capnproto and it is actually worse (all non-ascii characters change to octal notation)

[image]

not exactly sure why it is doing this, it is somewhere inside capnproto

an alternative would be to switch from `str()` to `to_dict()` and then use `pprint` to format it, but this approach leads to re-ordering of the keys and turning off all the sorting doesn't fix it (`logMonoTime` goes to the bottom)

adeebshihadeh, 2021-05-14:
This isn't really an actionable openpilot issue and personally don't think the format is a huge deal. We have a test to check for duplicate FW. Closing and will open an issue in xx to track this when we get some more time to look into it.

closed 2021-05-14 by adeebshihadeh